pozyxLabs/Pozyx-Arduino-library #2: Arduino API still detects 0 anchors even after issue #1

okachobi, 2016-02-20:
Even after adding a delay(10000) after setup, the pozyx arduino library is reporting 0 anchors when 4 are setup...

----------POZYX POSITIONING V1.0----------
NOTES:
- No parameters required.

- System will auto start calibration

- System will auto start positioning
----------POZYX POSITIONING V1.0----------

Performing auto anchor calibration:
1
list size: 0
Calibration result:
Anchors found: 0
Starting positioning:

RoboBil, 2016-03-01:
I've had the same issue. Waiting for the fix.

samuelvdv, 2016-03-02:
Thank you for the information, 
We are changing the auto calibration function to solve this and will update it soon.
In the mean time, you could add the devices manually if you know the coordinates:

Here is some sample code to add an anchor manually:

`device_coordinates_t anchor;  
 anchor.network_id = 0x666D;
 anchor.flag = 0x1; // indicate that it is an anchor. 
 anchor.pos.x = 0;
 anchor.pos.y = 0;
 anchor.pos.z = 3900;
 Pozyx.addDevice(anchor);`

Also, we'll be putting Arduino documentation online next week where more information about the functions and their parameters is provided.

Best!

RoboBil, 2016-03-02:
That did it. Just added the code below right after cleardevices line in the setup

Thanks

RoboBill

device_coordinates_t anchor1;
	anchor1.network_id = 0x6666;
	anchor1.flag = 0x1; // indicate that it is an anchor. 
	anchor1.pos.x = 0;
	anchor1.pos.y = 0;
	anchor1.pos.z = 1067;
	Pozyx.addDevice(anchor1);

	device_coordinates_t anchor2;
	anchor2.network_id = 0x6642;
	anchor2.flag = 0x1; // indicate that it is an anchor. 
	anchor2.pos.x = 8800;
	anchor2.pos.y = 0;
	anchor2.pos.z = 1067;
	Pozyx.addDevice(anchor2);
	device_coordinates_t anchor3;
	anchor3.network_id = 0x666B;
	anchor3.flag = 0x1; // indicate that it is an anchor. 
	anchor3.pos.x = 0;
	anchor3.pos.y = 3600;
	anchor3.pos.z = 1067;
	Pozyx.addDevice(anchor3);

	device_coordinates_t anchor4;
	anchor4.network_id = 0x6643;
	anchor4.flag = 0x1; // indicate that it is an anchor. 
	anchor4.pos.x = 8800;
	anchor4.pos.y = 3600;
	anchor4.pos.z = 1067;
	Pozyx.addDevice(anchor4);

SilkSama, 2016-06-03:
Hi, have you some informations about the date of the next release with the automatic calibrating fix ?
Thx