pozyxLabs/Pozyx-Arduino-library #3: Anchors arent sending anything (or the tag isn't receiving)

mgsfan91, 2016-03-22:
I'm using Arduino Uno and I've connected the Pozyx tag on it. The Pozyx.begin() function was fine (that means I've connected the tag to the Arduino correctly - introduction message was printed). The Pozyx.remoteRegWrite() is also fine - I can turn the leds off on all anchors I have at my disposal. But Pozyx.remoteRegRead() and ranging example fail. I can't get any response from any anchor. I've tried reading RX data registry on my tag after doing remoteRegRead() - it's empty. What can I try next?
Thx in advance for an answer! Best regards

samuelvdv, 2016-03-22:
Hello, 
are you using an arduino UNO R3?
Also, are the leds blinking on the anchors? (providing you did not turn them off). If the TX led is blinking on the anchor after doing Pozyx.remoteRegRead(), it means the anchor responded correctly and the problem is at the tag (maybe it's an interrupt problem).
Hope this helps, 

Please check [https://www.pozyx.io/Documentation/Tutorials/basics](https://www.pozyx.io/Documentation/Tutorials/basics) for a tutorial on all other things you can check for troubleshooting.

mgsfan91, 2016-03-23:
Hi Samuel,

thx for a prompt response.
Yes, Uno R3.
The leds are blinking, the LED1 every 2s, the other red led (the TX) more often.
The anchors are probably fine. On our Institute they managed to run the ranging and localization example, with the same  anchors as I am using, but with a different tag. So the tag is the problem. Should I try to update the tag?

Thx in advance for an answer!

P.S. Some other stuff I tried (just in case):

When doing the Device check example from the basics, I get:
**************************************************************************
who am i: 43
firmware version: 1
hardware version: 23
self test result: 111111
error: 0
**************************************************************************
which is fine. When doing the Remote device check from the basics, I get:
*************************************************************************
who am i: 0
firmware version: 0
hardware version: 0
self test result: 0
error: 0
*************************************************************************
The green led (RX) blinks on the anchor after the execution of this code (thx TX constantly blinks). When doing Network check I get:
*************************************************************************
Discovery found: 0 device(s).
Assertion failed in function : getDeviceIds
Filename: C:\Users\Ivan\Documents\Arduino\libraries\Pozyx-Arduino-library-master\Pozyx_lib.cpp
Line number: 1173
*************************************************************************
The green led also blinks here. 
Line 1173 is the assert assert(size > 0); (number of discovered devices is 0, so it fails).

When I check POZYX_UWB_CHANNEL of the tag, I get 5 (which it should be).
When I check POZYX_INT_MASK of the tag, I get 11111 (binary, means that all 5 interrupt sources are enabled).
When I put read of the POZYX_INT_STATUS in the loop() section, I ocassionally get the value 100 (means the new IMU measurement is ready, which is also fine) and mostly get all zeroes.

samuelvdv, 2016-03-23:
Hi, 
As this doesn't seem to be a library issue, it doesn't really belong on the github and I will continue the discussion further by email (from support@pozyx.io).
Best regards

closed 2016-03-23 by samuelvdv