pozyxLabs/Pozyx-Arduino-library #5: Smoothing positioning results

qbereau, 2016-06-28:
We were able to quite accurately 3D position our tag using your ready-to-localize sample & processing file. But it seems that while positioning is mostly accurate, it's hard to keep a steady position. We get a lot of noisy data.

Do you intend to implement some kind of kalman filter integrating the position & sensors data? Or maybe some other kind of filter?

Right now, what would you say is the best approach to keep an accurate position steady?

samuelvdv, 2016-07-09:
Hi,
currently the system only outputs 'raw' positioning data (without smoothing). We will add our filters that fuse UWB and IMU data in the firmware update scheduled at the end of the summer when we do a large update concerning positioning robustness. They are currently being tested in different scenarios while keeping the code & memory footprint small enough to fit on the pozyx microcontroller.

However, at this time, you could already use a lowpass filter to smooth the data. As you probably know, this is a commonly used filter which is simple to implement that removes any high frequent noise.

Best regards, 
Samuel

closed 2016-07-09 by samuelvdv

razackkpa, 2017-09-13:
Hi Samuel,
Is this filter integrated into the firmware completed.

tommitytom, 2017-10-09:
Also curious about this.  Do you have a usage example of the filters in action?  I notice the functionality has been added to the latest version of the library but the documentation seems to be copypasta from another function - and even making educated guesses on what the parameters might be I seem to be getting the same data out whatever I use!

I remember reading somewhere that the "strength" parameter is between 0-15, so I'm using something along the lines of this:
`Pozyx.setPositionFilter(FILTER_TYPE_MOVINGAVERAGE, 0xA);`
Is this correct?