pozyxLabs/Pozyx-Arduino-library #8: doDiscovery randomly fails

lpdunwell, 2016-07-04:
`status = Pozyx.doDiscovery(POZYX_DISCOVERY_ALL_DEVICES, 5, 10);`
randomly fails with status = 8.
In that case `Pozyx.getSystemError()` doesn't return anything at all.

lpdunwell, 2016-07-06:
well 8 seems to be a timeout. debugging errors is unreliable with no data coming out of getSystemError()

samuelvdv, 2016-07-09:
Hi, 
The doDiscovery() should give the error 0x0B (POZYX_ERROR_DISCOVERY) when it fails. This was not the case, rather you received a timeout. The doDiscovery waits for an interrupt that notifies the function that discovery is finished. I suspect that perhaps the interrupts are not configured properly (the POZYX_INT_MASK_FUNC should be set in the POZYX_INT_MASK register) such that the function doDiscovery times out, even though the discovery succeeded. 
I have fixed this in the library by temporarily switching to polling mode such that the function does not depend on the interrupt configuration. This is indeed better behavior, thank you for noticing this!
Samuel

lpdunwell, 2016-07-12:
I'm afraid it still fails with a timeout. Definitely less predictably, but still. Higher values for slots and duration seem to raise the chance for a fail.

lpdunwell, 2016-07-12:
Ignore the slots and duration part. It fails. Randomly. That's about all I know right now.

samuelvdv, 2016-07-12:
Hi Luke Patrick, 
you gave me a good hint there, the timeout happens when the function doesn't return anything after 100ms. How long it takes to de discovery depends on a number of things:

- slot duration in ms: T
- slots: M 
- number of devices: N

The tag will broadcast a signal and the other devices can answer, one in each slot, if no answer is received after M slots, the discovery stops. The slot duration determines how long the slot takes and depends on the UWB settings: depending on the UWB settings, wireless packets can take 100µs or multiple ms. 
The absolute minimum time for discovery is T*(N+M). I have changed the timeout in the doDiscovery such that it takes this into account. I hope this fixes your issue now.

closed 2016-07-12 by samuelvdv