pozyxLabs/Pozyx-Arduino-library #10: documentation incomplete + temperature bug

lpdunwell, 2016-07-05:
`int PozyxClass::getTemperature_c(float32_t *temperature, uint16_t remote_id)`

seems to return 0. Also the online documentation doesn't contain much.

lpdunwell, 2016-07-06:
on the website i see the temperature is unsupported in FW v0.9.
reading the FW version from the shield returns '1'.

samuelvdv, 2016-07-09:
Hi, 
This is an issue with the firmware version which has the value '1' for version v0.9. This should be '9'.
The next firmware version will follow the correct numbering format: v{MAJOR}.{MINOR} and also correctly fill the temperature register.

closed 2016-07-09 by samuelvdv

lpdunwell, 2016-07-12:
Great! thanks for the fixes! When will the fw be released?

samuelvdv, 2016-07-12:
Hi!
We hope this week but if not, then definitely still this month.
Thanks for all your comments by the way! We appreciate it.

lpdunwell, 2016-07-12:
Great!
Thanks. Just trying to help, as we basically have something relying on posyx and lose a lot of time if we need to reinvent the wheel... :D