pozyxLabs/Pozyx-Arduino-library #11: getDeviceIds randomly fails

lpdunwell, 2016-07-05:
`static int getDeviceIds(uint16_t devices[], int size = MAX_ANCHORS_IN_LIST, uint16_t remote_id = NULL);`
randomly fails, without any details when calling `Pozyx.getSystemError()`

samuelvdv, 2016-07-09:
Hi, 
I have added a check to the getDeviceIds() function and it will now return POZYX_FAILURE when it tries to read out more devices than available. This can also be verified by using getDeviceListSize() that returns the number of devices available.

I have also rewritten getTagIds() and getAnchorIds() to follow the same behavior.

I have also updated the pozyx firmware to give a system error if you try to read out more devices from 	POZYX_DEVICES_GETIDS than possible. However, the error can only be generated after the I2C is completed, because I2C reads one byte at a time. and the pozyx device only knows how many bytes were read at the end of the i2c transmission. This will be available in the next firmware release.

closed 2016-07-09 by samuelvdv

lpdunwell, 2016-07-12:
this needs to be reopened.
the `assert(size > 0);` still breaks it.

samuelvdv, 2016-07-12:
I am not sure how this breaks it? I have now removed the default value of the size argument. But supplying a size > 0 is necessary, hence the assert. The question if an assert is a good way of checking is another debate, as is noted in the other issue :)

lpdunwell, 2016-07-12:
okay I'll stfu about that for now. :D it just requires another check between scanning and getting anchor ids...

lpdunwell, 2016-07-12:
issue #2 might be related?