pozyxLabs/Pozyx-Arduino-library #13: setPositioningAnchorIds() doesn't work with doPositioning(...,POZYX_POS_ALG_LS)

Efreeto, 2016-07-05:
Please provide more documentation for setPositioningAnchorIds(). I intend to use Pozyx with multiple devices in a large area with many occluded locations.

I have set setSelectionOfAnchors() successfully, but it doesn't work with doPositioning(...,_POZYX_POS_ALG_LS_), while it seems to work with doPositioning(...,_POZYX_POS_ALG_UWB_ONLY_)

samuelvdv, 2016-07-12:
Hello, 
the setPositioningAnchorIds() allows you to manually specify which anchors to use for positioning. This can be useful when you have multiple anchors and don't want to use them all. 
Currently, the automatic anchor selection is available but naively selects the first 4 anchors in the device list because the dev kit and loc kit don't contain more than 4 anchors. I have added a note in our firmware todo list  to improve this for one of the next firmware updates.

Concerning your issue: could you share your code so I can reproduce your problem?

Best regards, 
Samuel

Efreeto, 2016-07-12:
Thank you for replying and responding with a follow-up task.
Unfortunately, I figured that anchors more than 4 are not well supported, so I had moved on to using my own selecting and positioning algorithms for quite some time. So I don't think it benefits you that I share my code from 7 days ago, which probably has other issues besides this one.

samuelvdv, 2016-07-13:
Ok, that is fine. 
I will close this issue now and once the functionality is included you will hear about it! :)
Best regards, 
Samuel

closed 2016-07-13 by samuelvdv