pozyxLabs/Pozyx-Arduino-library #16: Writing to status registers

lpdunwell, 2016-07-06:
Writing to status registers returns `POZYX_SUCCESS`. The data doesn't get written to the registers, though. I believe this should return a `POZYX_FAILURE`.

samuelvdv, 2016-07-09:
The function regWrite() in Pozyx_core.cpp now verifies if the register is writable.
If not, the function will return POZYX_FAILURE.
The same check has been added to regRead() and regFunction()
Thanks.

closed 2016-07-09 by samuelvdv

lpdunwell, 2016-07-12:
Thanks for the fix, but now error handling is totally broken, if not before anyway.

You're returning the same error as in general I2C-related errors.
`if(!IS_REG_READABLE(reg_address))
    return POZYX_FAILURE;`
vs.
```
n = Wire.endTransmission(false);    // hold the I2C-bus for a repeated start

  if (n != 0)
    return (POZYX_FAILURE);
```
and
```
if ( i != read_len){
    return (POZYX_FAILURE);
  }
```

This takes away the possibilities of any kind of useful error handling in live applications, as for "oh shit, the shield fell off, I better go into failure mode"

(actually I just verified this by unplugging the shield during a loop reading registers which are not readable anyway (0xC2 upwards). The Function `String PozyxClass::getSystemError(uint16_t remote_id)` in any case returned `POZYX_ERROR_NONE`)