pozyxLabs/Pozyx-Arduino-library #19: assertions

lpdunwell, 2016-07-07:
most of the functions in the arduino library have some kind of assertions.
now if this is a good thing or not, is a matter of personal opinion, but consider this:
```
<snip>
status = Pozyx.getDeviceListSize(&listsize);
...
status = Pozyx.getTagIds(anchors, listsize);
<snip>
```
if no tags are around, this will crash at runtime. crashing at runtime is bad.

also, may i suggest having a look at this?
https://www.softwariness.com/articles/assertions-in-cpp/

lpdunwell, 2016-07-13:
I'm really sorry for poking around in the assertions thing again, but I just begin to realize I have indeed a bit of a problem, as basically any assertion that fails will freeze our camera tracking and kill our fallback telemetry link. The problem is, I can't catch failed assertions when compiling, and I cannot react and handle them at runtime, as the assert function(s) halt the arduino. Of course I can modify the library, but at least I wanted to explain my exact problem.

samuelvdv, 2016-07-13:
Hi, 
I understand that asserts are not always the best choice. We decided to go for them because the Arduino library is mostly as an example of how to interact with the pozyx registers. Although, at this point the library proves to be more useful than this. I don't plan to remove all asserts overnight, but I'll take a look and see what better alternative to choose in the future.

For now, you can either use the NDEBUG define:
```C
#define NDEBUG 
```
to disable all asserts. You can also use:
```C
#define __ASSERT_USE_STDERR 
```
to give more information on a failed assert (this is takes a lot of RAM memory though).
Or you can overload the assert function, as it is defined as a weak function in the library. The following sketch shows this:

```C
#include <Pozyx.h>
#include <Pozyx_definitions.h>
#include <Wire.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
    
  Pozyx.begin();
  Serial.println("begin");
  
  // the following operation is illegal and will trigger an assert
  Pozyx.setTxPower(999.0f);  
  
  Serial.println("starting loop");
}

void loop() {
  // put your main code here, to run repeatedly:
}

void  __assert_pozyx (const char *__func, const char *__file, int __lineno)
{
    Serial.println("My custom assert does not abort operation");
}
```

closed 2016-07-13 by samuelvdv