pozyxLabs/Pozyx-Arduino-library #20: Return values

lpdunwell, 2016-07-12:
Generally, return values are:
`#define POZYX_FAILURE               0x0#define POZYX_SUCCESS               0x1 
#define POZYX_TIMEOUT               0x8`

which is different than the defines for `String PozyxClass::getSystemError(uint16_t remote_id)`:
`#define POZYX_ERROR_NONE                  0x00
#define POZYX_ERROR_I2C_WRITE             0x01  
...
#define POZYX_ERROR_STARTUP_BUSFAULT      0x08`

This should be unified, or solved differently than with defines.

lpdunwell, 2016-07-12:
also, there's a harmless typo in `String PozyxClass::getSystemError(uint16_t remote_id)`
```
default:
      return F("Unknonw error");
```

samuelvdv, 2016-07-12:
I have changed the minor typo and also altered the function to give a better error string when the pozyx device is not connected.
However, for the return values, I agree with you, but I have to think about how to change this. Simply changing POZYX_ERROR_NONE and POZYX_ERROR_I2C_WRITE would solve this but perhaps some people don't use the defines or have ported (parts of) the library and their code may fail after such an update. Any suggestions?

lpdunwell, 2016-07-12:
this indeed is a tough one. are there any major changes ahead in the fw that will break support?
also, while this is great if you basically watch the terminal for errors, for real applications a lengthy string as an error message just doesn't quite do it. I'd prefer to have details right when they fail. like forward the error code of the Wire functions. I mean, there's a lot more than just PASS and FAIL there to work with, and basically for the sw that's all there is right now... (I can't distinguish a bus fault from writing to a read only register without comparing strings)