pozyxLabs/Pozyx-Arduino-library #21: POZYX_CALIB_STATUS register

lpdunwell, 2016-07-12:
Actually I'm not entirely sure what to call this.
SYS Calib Status usually is not set (the rest is). But I actually had it the other way round.
Now, I cannot tell how severe this is, as I'm not entirely sure what to do with the information this register offers.

samuelvdv, 2016-07-12:
Hello, 
I have added more information about this mysterious register to the documentation.
It can be found here: 
https://www.pozyx.io/Documentation/Datasheet/RegisterOverview
Hopefully it is more clear now.

closed 2016-07-12 by samuelvdv

lpdunwell, 2016-07-13:
Thanks.
I just booted the arduino and got:

> Calibration Status:
> MAG: fail
> ACC: fail
> GYR: ok
> SYS: fail

I made a couple of movements as suggested, reset the arduino and got:

> Calibration Status:
> MAG: ok
> ACC: fail
> GYR: ok
> SYS: ok

Which I believe should not be possible? SYS should fail, as ACC is not calibrated?

lpdunwell, 2016-07-13:
I actually managed to get ACC calibrated, but SYS comes on too early

samuelvdv, 2016-07-13:
The system calibration indicates if the sensors are **sufficiently** calibrated to compute the orientation of the device by fusing all the data from the motion sensors. 

The accelerometer already has an internal initial calibration which is mostly sufficient in combination with the gyroscope and magnetometer, hence, even when the accelerometer isn't completely calibrated, it is still sufficient to obtain the orientation.

When the magnetometer is not calibrated, or when the sensor is close to an external magnetic field, it is impossible to accurately measure the absolute heading. You can verify this if you continuously print out the calibration status: bringing some metal or a magnet close to the pozyx board (in fact the motion sensor) will make SYS and MAG status go down. When it can be recognized that this occurs the magnetomer will not be used to obtain the orientation of the device and you will see that after a while the heading will drift away.

To conclude, the sys calib status gives you an indication whether the orientation data is reliable.

Please check out [this processing sketch for Pozyx that shows all the orientation data](https://github.com/pozyxLabs/Pozyx-processing)

I'm considering making a video about this actually. We'll see :)