pozyxLabs/Pozyx-Arduino-library #28: getDeviceRangeInfo messes up big time

lpdunwell, 2016-07-18:
setup: 5 anchors

code (the important part):
```
coordinates_t position;  
  Pozyx.doPositioning(&position, POZYX_2_5D, 1000, POZYX_POS_ALG_LS);
  
  for(int i=0; i < listsize; i++)
  {
    device_range_t range;
    Pozyx.getDeviceRangeInfo(anchors[i], &range);
    Serial.print(anchors[i], HEX);
    Serial.print(": ");
    Serial.print(range.distance);  
    Serial.print(", ");
    Serial.println(range.RSS); 
  }
  Serial.println();
  Serial.println();
  
// following part straight from the example, as it may be a timing issue on the stm32
  for (int i=0; i < listsize; i++){
    device_range_t range;
    Pozyx.getDeviceRangeInfo(anchors[i], &range);
    Serial.print(",");
    Serial.print(range.distance);  
    Serial.print(",");
    Serial.print(range.RSS); 
  }
```

the output:
```
603E: 18194, -81
601F: 5595, -81
6018: 12998, -82
6047: 447, -85
6048: 447, -85

,18194,-81,5595,-81,12998,-82,447,-85,1728816549,2050
```
as you might suspect reading this issue report, I spent some time wondering why Anchors 4 and 5 have the same data. The problem you see here actually is far worse than just overwriting A5 data with that of A4...

What's the state of the new FW? IIRC you said something about last week? The longer I'm forced to fiddle around with V0.9, the more weird-as-hell issues will be reported. Now I don't know what is, what isn't and what will be fixed, but I can imagine there's a lot of change going to happen, so I am a bit lost in the woods.

Cheers

lpdunwell, 2016-07-18:
another try:

> 603E: 18180, -81
> 601F: 5488, -81
> 6048: 5337, -82
> 6018: 12998, -82
> 6047: 78534, 0
> 
> 
> ,18180,-81,5488,-81,5337,-82,12998,-82,12998,-82

lpdunwell, 2016-07-18:
it's packet corruption in the air.
the failure rate is too high to be practical. even in sub 5 meter distances

samuelvdv, 2016-07-18:
Hi, 
what you are seeing here is in fact expected behavior: 
the ```doPositioning()``` function will perform ranging with the number of anchors specified by the register POZYX_POS_NUM_ANCHORS, which has the value of 4 by default. It is not packet corruption, it is simply not configured to do ranging.
In your code, you should check if ```Pozyx.getDeviceRangeInfo(anchors[i], &range);``` returns success or not. When no success the data is invalid. It turns out to be the old memory which may be unwanted, I have added it to the firmware issue track to change this in zeros. Concerning the firmware, we want to test everything thoroughly before releasing, which we have done for most parts now.

The following code shows how the getDeviceRangeInfo works:
```C
#include <Pozyx.h>
#include <Pozyx_definitions.h>
#include <Wire.h>

void setup() {
  Serial.begin(115200);    
  Pozyx.begin(false);
  
  int status = Pozyx.doDiscovery();
  Serial.println(status);
   
  uint8_t list_size = 0;
  status = Pozyx.getDeviceListSize(&list_size);
  Serial.print("Number of pozyx devices discovered: ");
  Serial.println(list_size);
  
  uint16_t devices[list_size];
  status = Pozyx.getDeviceIds(devices, list_size);
  
  int i;
  for(i=0; i<list_size; i++)
  {
    device_range_t range;
    range.distance = 0;
    range.RSS = 0;
    status = Pozyx.doRanging(devices[i], &range);
    Serial.print("Ranging status: ");
    Serial.println(status);      
    Serial.print("Distance (mm): ");
    Serial.println(range.distance);
    Serial.print("Signal strength (dBm): ");
    Serial.println(range.RSS);
  }
  
  // now let's try to grab ranging information from an non-existing node (0xFFFF) -> make sure this is not one of your anchor ids
  device_range_t range;
  status = Pozyx.getDeviceRangeInfo(0xFFFF, &range);
  Serial.print("status: ");
  Serial.println(status);      
  Serial.print("Distance (mm): ");
  Serial.println(range.distance);
  Serial.print("Signal strength (dBm): ");
  Serial.println(range.RSS);  
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

The output I get is:
```
1
Number of pozyx devices discovered: 4
Ranging status: 1
Distance (mm): 2772
Signal strength (dBm): -80
Ranging status: 1
Distance (mm): 6818
Signal strength (dBm): -88
Ranging status: 1
Distance (mm): 8623
Signal strength (dBm): -92
Ranging status: 1
Distance (mm): 10307
Signal strength (dBm): -98
status: 0
Distance (mm): 10307
Signal strength (dBm): -98
```

closed 2016-07-18 by samuelvdv

lpdunwell, 2016-07-18:
okay, but how do I work with this simple check timing out or losing a packet in 90% of attempts?

lpdunwell, 2016-07-18:
more importantly, how am I supposed to exclude any possible extraneous anchors in setting up the system? if only by chance someone within reach uses Pozyx too, and an anchor gets picked up additionally to my 4, that's going to break the system isn't it?