pozyxLabs/Pozyx-Arduino-library #31: pozyx definition versioning && POZYX_FLASH_SAVE

makrebs, 2016-08-23:
The pozyx documentation website states some new functions as [POZYX_FLASH_SAVE](https://www.pozyx.io/Documentation/Datasheet/RegisterOverview#POZYX_FLASH_SAVE) which are also listed in the [v1.0 register header](https://www.pozyx.io/Documentation/Datasheet/Registerheader/v1.0).
The arduino library however is missing those, although [this commit](https://github.com/pozyxLabs/Pozyx-Arduino-library/commit/5282db6a5f4e429ed6f7c8626514d572db6c3ed8) states the lib has been updated to v1.0.1.
How can I use the POZYX_FLASH_SAVE function (a very useful feature btw).
+Matthias

samuelvdv, 2016-08-23:
Hi,
I have just uploaded the new Arduino library that includes these functionalities.
It is true that the Arduino library version is confusing with the Pozyx firmware version. Now, the releases are relabeled to match the firmware version. I hope this is more clear now.

Note that the new pozyx firmware must be installed first to be able to use these new functions. You will receive this by email later today or tomorrow.

Best regards
Samuel

closed 2016-08-23 by samuelvdv

makrebs, 2016-08-23:
Great. Thanks for the library udate.

Yes please send me the latest firmware such that we can use the new
functionalities.

Best,
Matthias

-- 

*Matthias Krebs*CTO, Co-Founder

Mobile: +41 79 431 8271
Office: +41 44 633 8216
Mail: matthias@ <matthias@aerotainmentlabs.com>aerotain.com
Web: www.aerotain.com

AEROTAIN
Leonhardstrasse 27
CH-8092 Zurich
Switzerland