pozyxLabs/Pozyx-Arduino-library #34: begin doesn't always work

lpdunwell, 2016-08-26:
sometimes it just fails.
i would add details, if i had any.

samuelvdv, 2016-09-01:
Hello,

I would assume you can give some more details anyhow. What Arduino are you using and what do you mean with 'fails'? At what point does it fail? 

Best regards,
Samuel

lpdunwell, 2016-09-01:
if(Pozyx.begin(true) == POZYX_FAILURE)

it happens. randomly.
it's a mega2560.

samuelvdv, 2016-09-02:
Hi,
I have tested it with the MEGA2560 and everything works fine here. So it cannot be a library issue.

* does it happen on all sketches? Also the example sketches included in the library?
* Are the LED's blinking on the Pozyx device? (red one slowly and green Rx fast, The other leds should not be lit). Simply powering the device should make these LEDS blink
* Is the jumper present on the pins BOOT0? If not, the pozyx device will not start
* Is the firmware correctly updated?
* In the Arduino IDE, have you selected 'Arduino Mega2560' as the target board (under tools)?

lpdunwell, 2016-09-02:
we're talking about countless hours of automated testing of switching on
power and setting up a network.

i'm afraid, your answer sounds to me as if you're relying on manual tests.
what's your testbench setup and parameters?

cheers

---
Luke P. Dunwell

samuelvdv, 2016-09-02:
Hello, 

we have written numerous unit tests to test the functionality of the Pozyx device. Only a few have been released here (see the folder in the library), because other require a fixed setup with more control over the test environment. However, this is still a work in progress, and we don't claim it to be perfect: we are constantly working on improving this. Note that during production we also perform the selftest to see if each component on the board is working correctly.

The Arduino library is now divided in two parts, one that is platform dependent and another that is platform independent. The platform dependent test check if i2c communication and interrupts all work as expected. If this tests succeeds on a given Arduino platform, we can conclude that the Arduino platform is compatible with Pozyx.
The platform independent tests are there to test the Pozyx functionalities. We do not test these on all different Arduino platforms as they are platform independent. 

Considering this github issue, I have executed the platform dependent unit tests (which includes the begin() function) on the Arduino Mega with a Pozyx tag (firmware v1.0) to verify that the library is compatible with the Mega. This test was successful as you can see in the image below. 

[image]

Because of this, I believe that the problem does not lie in software, but rather something else.

closed 2016-12-08 by laurentva