pozyxLabs/Pozyx-Arduino-library #35: begin doesn't work at all on new tags

Efreeto, 2016-09-02:
I have a similar issue as https://github.com/pozyxLabs/Pozyx-Arduino-library/issues/34, but my issue is a little different. I also emailed Vadim about this a few days ago. (no response from him yet)

We purchased the Pozyx kits on two separate orders. While the tags from my older order work well all the time, but the tags we received recently fail at begin() all the time.

Here is the code I used:
```
void setup(){
  Serial.begin(115200);
  
  if(Pozyx.begin(true) == POZYX_FAILURE){
    Serial.println(F("ERROR: Unable to connect to POZYX shield"));
    Serial.println(F("Reset required"));
    delay(100);
    abort();
  }

  Serial.println(F("----------POZYX POSITIONING V1.0----------"));
  Serial.println(F("NOTES:"));
  Serial.println(F("- No parameters required."));
}
```
And this is the serial output
```
Pozyx Shield
------------
WhoAmI: 0x43
SW ver.: 1
HW ver.: 2
selftest: 0b111111
ERROR: Unable to connect to POZYX shield
Reset required
```
I used the same Arduino Uno on both sets of tags.

samuelvdv, 2016-09-02:
Hello,

it seems like the firmware was not updated. The new firmware has been sent by email together with instructions on how to update it. If you haven't received it, please let us know. It is required to update the firmware on all devices. If you don't want to update the new firmware, it is still possible to download the Arduino library compatible with the old firmware here: https://github.com/pozyxLabs/Pozyx-Arduino-library/releases

The firmware version shown in the begin() function is in unformatted so the value 1 does not mean fw v1.0. This is quite confusing and I have fixed this in the library.

Best regards,

Samuel

closed 2016-09-02 by samuelvdv

Efreeto, 2016-09-02:
Thank you so much Samuel!
I also received a solution mail from Laurent just now to on this issue. I will try the new firmware, and I'll let Laurent know from now.