pozyxLabs/Pozyx-Arduino-library #36: How to reconfigure the network id permanently?

Efreeto, 2016-09-02:
We purchased a lot of Pozyx tags and anchors, and we found some of them have duplicate network ids.
It seems setNetworkId() could be used for the trick, but I cannot get it to work. Plus, the documentation says it will be only temporary until "the system is reset", whatever that means.

Is there a way to reconfigure the network id permanently? If not, can you describe how to use setNetworkId()? I sent an email to Vadim with the code I was trying to use setNetworkId().

samuelvdv, 2016-09-02:
Hello,

changing the network id can be done quite easily with these lines of code:

```C
uint16_t new_network_id = 0x1111;  // example network id
result = Pozyx.setNetworkId(new_network_id );
```

However, this will not change it permanently, like any other register, it will return to its default value after reset. To counter this it is necessary to save the register value to flash, this can be done with the following command:

```C
// save to flash  
uint8_t reg = POZYX_NETWORK_ID;         // this says that we want to save the network id 
result = Pozyx.saveConfiguration(POZYX_FLASH_REGS, &reg, 1);
```

Now, the network ID will remain the same even after reset. If you call the function ``Pozyx.clearConfiguration() ``, all registers will be reset to their default value, including the network id.

Too save you the trouble of writing any code, here is a script that changes the network ID of the tag connected to the Arduino. The script also performs some extra verification:

```C
#include <Pozyx.h>
#include <Pozyx_definitions.h>
#include <Wire.h>

// fill in the new network id here
uint16_t network_id = 0x1112;

void setup() {
  Serial.begin(115200);
  while(!Serial); // for the Arduino Leonardo/Micro only
  
  Pozyx.begin();
  
  Serial.println(F("Changing Network ID"));  
  Serial.println(F("---------------------------------------------------------\n")); 
  
  // put your setup code here, to run once:
  uint8_t result, num_regs;   
  
  uint16_t network_id_old = 0;
  result = Pozyx.getNetworkId(&network_id_old);
  Serial.print("Current network id: 0x");
  Serial.println(network_id_old, HEX);
  
  if(network_id_old == network_id)
  {
    Serial.println("Not changing anything");
  }else
  {  
    // if you want: you can perform a memory clear
    //result = Pozyx.clearConfiguration(); 
        
    // write a new value in the register
    result = Pozyx.setNetworkId(network_id);
    delay(100);
    
    // save to flash  
    uint8_t reg = POZYX_NETWORK_ID;
    result = Pozyx.saveConfiguration(POZYX_FLASH_REGS, &reg, 1);
    if(result != POZYX_SUCCESS)
    {
      Serial.println(result);
      Serial.println(POZYX_SUCCESS);
      Serial.println("could not save the network ID");
    }
    
    result = Pozyx.isRegisterSaved(POZYX_NETWORK_ID);
    if(result != POZYX_SUCCESS) {
      Serial.println("Channel was not saved.");
      Serial.println(result);
    }  
    
    // reset the Pozyx device to verify if it worked
    Pozyx.resetSystem();
    
    network_id = 0;
    result = Pozyx.getNetworkId(&network_id);
    Serial.print("Network id after reset: 0x");
    Serial.println(network_id, HEX);
  }
      
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Hope this helps,
if you want to change the network id remotely, I can write you a little script for that as well

Cheers, Samuel

closed 2016-09-02 by samuelvdv

Efreeto, 2016-09-03:
Hello Samuel,
Thank you for replying so quickly and posting your code!

Unfortunately, when I ran the code above I got the following error:

```
---------------------------------------------------------

Current network id: 0x606F
8
1
could not save the network ID
ChannåChanging Network ID
---------------------------------------------------------

Current network id: 0x606F
8
1
could not save the network ID
ChannåChanging Network ID
```

As you can see it fails saveConfiguration(), and the original Id 0x606F is still shown in the next iteration. By the way, I don't know why it's going through a loop if everything is in the setup() function.

Efreeto, 2016-09-06:
Samuel, the code is working after I upgraded the firmware to 1.0. I verified the new network id is saved even after unplugging the device.
Thank you very much for your reply and especially for sharing your code.

samuelvdv, 2016-09-06:
All right, i am happy to hear it is all working well now!