pozyxLabs/Pozyx-Arduino-library #37: ready_for_localize of doAnchorCalibration

qdzhaozx, 2016-09-02:
To call the function doAnchorCalibration,the problem happens in step to add devices by dimension be POZYX_2_5_D,the function returns zero.
__________________________________________________________________________________________________________
  // in 2.5D mode, we must supply the heights of all the anchors
  if(dimension == POZYX_2_5D){
    device_coordinates_t anchor;
    for(i=0; i< num_anchors; i++){
      anchor.network_id = anchors[i];
      anchor.flag = 0x1; 
      anchor.pos.x = 0;
      anchor.pos.y = 0;
      anchor.pos.z = heights[i];
      status = Pozyx.addDevice(anchor);
	  delay(500);
    }

qdzhaozx, 2016-09-02:
latest news!
I test the function "remoteRegRead" to read 4 anchors ID and SW versions ,the result is normal .Print log is like follows:
**get anchor informations: 
get remote status 1
who am i: 43
firmware version: 1
hardware version: 2**

So why not work of "addDevice"??

And function "clearDevices" is also normal!

samuelvdv, 2016-09-02:
Hello,

From your information it is clear the the devices do not have the latest firmware. 
It is strongly recommended to upgrade to the latest firmware on all devices.

If you do not want to upgrade the firmware for some reason, you can still use the older library version found here: https://github.com/pozyxLabs/Pozyx-Arduino-library/releases

closed 2016-09-02 by samuelvdv