pozyxLabs/Pozyx-Arduino-library #44: SPI bus to access DW1000's registers

fdespaux, 2017-01-12:
Hi,
I have one question regarding the DW1000.
Normally, we should be able to communicate with the transceiver by means of the SPI bus controller which allow us to access to DW1000's registers in order to obtain several parameters.

Is there any way to do so with tags and anchors ? I suppose you do this for setting parameters such as preamble length, channel, bitrate, etc.

Thank you,

François

samuelvdv, 2017-01-13:
Hello,
the Pozyx tags (and anchor) has its own processor on the board that is internally connected to the DW1000 through SPI. With the Arduino it is not possible to connect directly to the DW1000, only to the on-board Pozyx processor which supports i2c. You can still set several UWB related settings (the ones you mention) by accessing the Pozyx registers. You can find Arduino functions that do this here: https://www.pozyx.io/Documentation/Datasheet/arduino
Please check out this tutorial if you want to change the UWB parameters:
https://www.pozyx.io/Documentation/Tutorials/uwb_settings

Best regards,
Samuel

closed 2017-01-13 by samuelvdv

fdespaux, 2017-01-13:
Hi Samuel,
Thank you for your answer. 
I will then change the answer. Is there a way (by means of the Pozyx registers) to obtain some other information such as the timestamp from last message sent or last message received ?
I know that DW1000 allows us to obtain also the skew between clock's of both sender and receiver. Is there a way to obtain this information ?

Thanks,
François

samuelvdv, 2017-01-18:
Hello,
this functionality is not available. The timestamps are internally used during the ranging protocol but they are not saved in the registers. What exactly do you want to do?

Best,
Samuel

fdespaux, 2017-01-19:
Hello Samuel,

Thanks for your answer. What I want to do is to develop a localisation algorithm based on TDoA (Time difference of arrival). For doing this, it is necessary to obtain the timestamps from the moment a message coming from the tag, arrives to each of the anchors. After that, anchors exchange these timestamps in order to localise the tag (by means of an intersection of hyperbolas).

François

samuelvdv, 2017-01-19:
Hello François,
For the two-way ranging we are using a ranging algorithm that isn't affected by the clock offset of the devices. Because of this we can use a 15ppm crystal on the boards (tuned to about 2ppm). However, for TDOA all your anchors must be properly synchronized, and the more precise your crystal (and more expensive) the less frequent you have to perform synchronization. So with the current boards you will have to perform synchronization more frequently. Note that on the Decawave evaluation kits the crystal is even worse.

I will check if we can add a timestamp register in the next firmware release. However, just know that the boards were designed with two-way ranging in mind. 
We are currently also working ourselves on a TDOA implementation which has different hardware.

Best,
Samuel

fdespaux, 2017-01-19:
Hi Samuel,

Ok, thanks and that would be nice if we can access to timestamps through register in next firmware. In fact, not only this value, but also some other values from the dw1000 register.
Yes, I know the synchronisation problem you mentioned since last year, as part of a postdoc position I had, I have been working in synchronisation over uwb for tdoa-based localisation (results that I will be publishing soon).
Thanks again for your time.

François Despaux