pozyxLabs/Pozyx-Arduino-library #45: POZYX_OPERATION_MODE register values backwards

jake-magee, 2017-01-12:
My anchor boards (that do not have a jumper on the T/A pins) appear to be reporting "01" for the POZYX_OPERATION_MODE and the arduino shields (that do have a jumper on the T/A pins) appear to be reporting "00".  Based on the documentation and arduino/python library, this seem backwards.

samuelvdv, 2017-01-13:
Hello,
this seems to be wrong in the documentation. It should be fixed now here:
https://www.pozyx.io/Documentation/Datasheet/RegisterOverview#POZYX_OPERATION_MODE

Unless you saw this documentation somewhere else?

Thanks!

closed 2017-01-13 by samuelvdv

jake-magee, 2017-01-13:
Thanks for the clarification.  I must have got the values swapped in my head when I double checked with the libraries because they are infact correct.

jake-magee, 2017-01-13:
Yes, the values are also backwards in the arduino and python libraries.
https://github.com/pozyxLabs/Pozyx-Arduino-library/blob/8473bdc1332faf25dcbda78f241e7007f4bd456b/Pozyx_definitions.h#L223-L224
https://github.com/pozyxLabs/Pozyx-Python-library/blob/35cc3fef0da5f374b98bb2fa7f97050e750d8673/pypozyx/definitions/constants.py#L38-L39