pozyxLabs/Pozyx-Arduino-library #46: SYS Calibration status unstable

jake-magee, 2017-01-12:
I can eventually get the calibration status register to report 0xFF.  However, after small movements the register will move between 0xBF, 0x7F, and even 0x3F (the SYS portion is changing).  Is this expected?  I could understand the SYS portion changing if one of the other sensors were also changing.

samuelvdv, 2017-01-13:
Hello, the accelerometer, gyroscope and magnetometer (al on the Bosch sensor) perform on the fly calibration. The calibration status register both indicates if the calibration if successful and if the sensor is being used in the fusion algorithm. For example, if you would bring a magnetic device close to the pozyx board, the sensor will detect this anomoly and not use the magnetic sensor to compute the 3D orientation. This will be seen in the calibration status register (the two bits for the magnetometer and  two bits for the system)

closed 2017-01-13 by samuelvdv

jake-magee, 2017-01-13:
I'm still a little confused.  If the calibration status register reports 0x3F, should I consider the tag calibrated (since the GYR, ACC, and MAG are reporting as calibrated)?  I'm thinking that my assumption is wrong about the SYS status... I thought it was simply a reflection (bitwise AND) of the other 3 statuses.  Can you clarify what the SYS status represents?

samuelvdv, 2017-01-18:
The SYS status is not simply an AND of the other statuses. It is more an indication if all sensors are being used by the sensor fusion algorithm. The calibration status is a result of the BNO055 chip and is computed by the chip. Unfortunately Bosch doesn't provide any more information about this either. We have contacted them but for now we are without definite answer.