pozyxLabs/Pozyx-Arduino-library #47: POZYX_ERROR_FLASH_INVALID on power up

jake-magee, 2017-01-12:
I seem to get the FLASH_INVALID error every time I power up my tag.  It doesn't appear to matter how long I give the device to power up, and after reading it the first time I never seem to see it again.

Also, does reading the ERRORCODE register clear the ERR bit in the INT_STATUS register?

samuelvdv, 2017-01-13:
Hello,

the FLASH_INVALID error can be ignored, on startup it will give this error when nothing is saved in the flash memory. In this case, the random bits in the memory do not pass the checksum which causes this 'error' notification.

Reading the ERRCODE doesn't clear anything. Reacing the INT_STATUS however, clears the int_status register.

closed 2017-01-13 by samuelvdv

jake-magee, 2017-01-17:
For anyone else that runs into this, it appears that doing a POZYX_FLASH_RESET will rewrite the checksum correctly and fix this issue.  A POZYX_FLASH_SAVE call might also fix it if you don't want to reset the entire FLASH.  In my case, POZYX_FLASH_DETAILS reported that everything was already zeroed out, so I just used a FLASH_RESET call.