pozyxLabs/Pozyx-Arduino-library #52: Number of supported anchors

jakemagee, 2017-01-26:
How many anchors are supported by Pozyx?  In the register overview documentation, POZYX_POS_SET_ANCHOR_IDS references 10 while POZYX_POS_NUM_ANCHORS references 15.

samuelvdv, 2017-01-26:
Hi,
It can use up to 15 anchors for positioning when selecting them automatically. However, if you select them manually (as opposed to automatic) it is only 10. There is not really a reason for this (it's historically). We will probably make this equal in the next firmware version.

Best,
Samuel

closed 2017-01-26 by samuelvdv

jakemagee, 2017-01-26:
I also forgot to mention, the POZYX_POS_SET_ANCHOR_IDS reports to take 32 param bytes, but the input parameters table only documents 20 bytes (in the register overview documentation).

CERVERO1979, 2019-01-23:
Hi.

Is it possible to use 15 anchors in manual mode today?
Thank you.