pozyxLabs/Pozyx-Arduino-library #55: Which algorithms are implemented

AlexisTM, 2017-03-16:
Dear,

There are 3 algorithms in the definitions. 

Which of those are implemented?

* POZYX_POS_ALG_UWB_ONLY = 0
* POZYX_POS_ALG_TRACKING = 1
* POZYX_POS_ALG_LS = 2

samuelvdv, 2017-03-16:
Hello,

The POZYX_POS_ALG_UWB_ONLY and POZYX_POS_ALG_LS are currently available in both 2D and 3D. The POZYX_POS_ALG_TRACKING is currently under test and will arrive in the next firmware update.

Best Regards,

closed 2017-03-16 by samuelvdv

sveinolavhegerland, 2018-02-22:
Hello,

what sort of algorithm is POZYX_POS_ALG_UWB_ONLY? Is it some kind of Linear Least Squares (LLS), Non LLS, Cayley-menger etc. ?  The same for POZYX_POS_ALG_LS = 2??

Thanks in advance!

AlexisTM, 2018-02-22:
* POZYX_POS_ALG_UWB_ONLY: No post processing
* POZYX_POS_ALG_TRACKING: Using IMU
* POZYX_POS_ALG_LS: Least squares