pozyxLabs/Pozyx-Arduino-library #61: Heading angle is inconsistent between tags

AlexisTM, 2017-05-10:
Dear, 

The heading angle is inconsistent between tags. Why the heading isn't pointing North? It looks almost random when I reboot the Pozyx tags.

How can we be sure all the tags have the same reference?

Also, when turning the device of 90 degrees (not in the hands), it sometimes says 110°, sometimes 70°. 

How can we calibrate the IMU?

AlexisTM, 2017-05-12:
The issue is most probably coming from the fact that I am doing tests in a building with metal ceiling.

Is there a way to be sure we have the same 0° yaw?

samuelvdv, 2017-05-12:
Hello,

the orientation is measured by the Bosch BNO055 IMU (acc + magn + gyro). In order to get an absolute orientation, the IMU uses the magnetometer as a compass to find the magnetic north. However, like you mention, the presence of metal can make it impossible for the magnetometer to determine the magnetic north. Unfortunately there is no way around this. This also impacts the issue you have with turning the device, as the magnetometer is of no help. 
Furthermore, the IMU requires some time to be perfectly calibrated. However, for the best results this requires the devices to be moved around (similarly to how you are required to 'calibrate' the compass on an iPhone). If the device is not fully calibrated the results may be less reliable. You can see if the device is calibrated using the register POZYX_CALIB_STATUS on the pozyx device.

hope this helps.

closed 2017-05-12 by samuelvdv