pozyxLabs/Pozyx-Arduino-library #68: Using pozyx as sensor for autonomous drone

dirtygreenppt, 2018-01-04:
Hi. I’m facing issues when doing my project. I tried to use pozyx tag as sensor and followed the instructions on ardupilot however I cannot find the BCN parameter on Q ground control. Wonder anyone using pozyx as sensor for drone? And how you overcome this problem? 
Thanks!

RoboBil, 2018-01-04:
I'm using the POZYX and the **latest version** of the Michael Oborne Mission Planner for my rover.

dirtygreenppt, 2018-01-04:
Hi RoboBill,

In your project, Pozyx can communicate with Mission planner as ground
station? I’m doing an indoor autonomous drone project and having issue on
sensor( which is pozyx), do you find any data or website that shows how to
use pozyx as sensor?
Much thanks!

Yan Ming

dirtygreenppt, 2018-01-04:
@RoboBil
Hi RoboBill,

In your project, Pozyx can communicate with Mission planner as ground station? I’m doing an indoor autonomous drone project and having issue on sensor( which is pozyx), do you find any data or website that shows how to use pozyx as sensor? 
Much thanks!

RoboBil, 2018-01-04:
I'm using POZYX for guiding my rover from waypoint to waypoint. I follow the instructions given here: [http://ardupilot.org/copter/docs/common-pozyx.html]. I do not know how, but it seems to work.

dirtygreenppt, 2018-01-04:
I see, thank you so much!

dirtygreenppt, 2018-01-08:
@RoboBil
Hi, RboBil, may I ask did you add any pozyx information in your code?

RoboBil, 2018-01-08:
Nope, because I'm not a programmer.

dirtygreenppt, 2018-01-10:
Hi RoboBil, sorry that I keep asking you for help😅. Do you know are the
coordinates sent through ground station? I am facing this problem, when I
mount the pozyx on the drone, the serial moniter stop displaying the
coordinates of pozyx.

RoboBil, 2018-01-10:
sorry sounds like a programming question which is way above my skill set

avrilynnyap, 2018-01-14:
Hi robobil, I am also using Pozyx for my school project. Im facing some issues regarding my ground station. Is it possible for you to take screenshots of your mission planner when doing the tutorial as posted on the website? Thank you so much.

RoboBil, 2018-01-14:
Sorry, I'm NOT a programmer and I'm having my own troubles with the Pixhawk/MP due to my own self inflicted "wounds".

AlexisTM, 2018-02-22:
You can use the pozyx by reading; on the drone, the data. Either Arduino and Python will work. 

Note that you have to forward the data to the flight controller. 

2 solutions for that through MAVLink:
* PX4 Firmware: use MOCAP
* Ardupilot or PX4: Fake GPS data