pozyxLabs/Pozyx-Arduino-library #70: Compile Error

eren23, 2018-02-26:
Hi I was trying to compile ready to range code but that was the output after i compile file.

C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx_lib.cpp:783:5: error: prototype for 'int PozyxClass::doPositioning(coordinates_t*, uint8_t, int32_t, uint8_t)' does not match any in class 'PozyxClass'

 int PozyxClass::doPositioning(coordinates_t *position, uint8_t dimension, int32_t height, uint8_t algorithm)

     ^

In file included from C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx_lib.cpp:8:0:

C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx.h:1633:16: error: candidate is: static int PozyxClass::doPositioning(coordinates_t*, int32_t)

     static int doPositioning(coordinates_t *position, int32_t height = 0);

                ^

C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx_lib.cpp:851:5: error: prototype for 'int PozyxClass::doRemotePositioning(uint16_t, coordinates_t*, uint8_t, int32_t, uint8_t)' does not match any in class 'PozyxClass'

 int PozyxClass::doRemotePositioning(uint16_t remote_id, coordinates_t *coordinates, uint8_t dimension, int32_t height, uint8_t algorithm)

     ^

In file included from C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx_lib.cpp:8:0:

C:\Users\user\Desktop\arduino-1.8.5\libraries\Pozyx-Arduino-library-master\Pozyx.h:1653:16: error: candidate is: static int PozyxClass::doRemotePositioning(uint16_t, coordinates_t*, int32_t)

     static int doRemotePositioning(uint16_t remote_id, coordinates_t *coordinates, int32_t height = 0);

                ^

exit status 1

Thanks in advance for any help.

mehmetyldz87, 2018-03-02:
Hi @eren23  
We have same issue. You can solve temporarily by dropping the library version 1.1.3  to 1.1.2.  Pozyx support team tadvise us like that, will soon fix the problem.

Regards

laurentva, 2018-03-05:
I've now fixed this! My apologies, I forgot to change the function signature in the cpp file after changing the signature in the header.

closed 2018-03-05 by laurentva