pozyxLabs/Pozyx-Arduino-library #72: Useful sketches not shown in Arduino Create

laurentva, 2018-03-30:
Only examples are shown in the Arduino Create web editor. Moving useful to examples and making new subdivisions would make the library fully cloud compatible.

pieter-pozyx, 2018-03-30:
I will take a look at this next week

per1234, 2018-03-30:
If you want to organize the example sketches you can have subfolders under the examples folder, which are reflected in the menu:
[image]
The same will be done in the standard Arduino IDE's **File > Examples** menu so it benefits users of either program.

laurentva, 2018-03-30:
Hey Per, 

That's great! In the past we didn't need to put everything under a single folder, but I prefer it this way as well. Great move by Arduino :smile: