pozyxLabs/Pozyx-Arduino-library #74: POZYX_ERROR_UNRECDEV

lpdunwell, 2018-05-04:
the STM32 is trying to communicate with the DW1000, LEDs 1-4 are active, LEDs 1 and 4 go off for ~125us maybe once per second (seems to happen while the STM32 tries to communicate).
There is activity on all the SPI signals, supply voltages seem ok.
I tried reflashing FW v1.3, which did not help. With v1.1 i don't even get a VCP upon USB connection.
this is a v1.3 anchor.

bdz69, 2018-05-04:
I had the same problem. I was unable to revert to an earlier firmware version with the same symptoms you describe. I wrote to support@pozyx.io and the sent me a special version of the 1.1 firmware that would reset the board. I used my st-link v2 programmer and completely wiped the chip with a full erase. Afterwords I had to use an Arduino Uno and use the pozyx-change-network-id sketch to give the board an id; It was reset to 0x0000. After that I was able to reprogram the chip normally with the proper UWB settings.

lpdunwell, 2018-05-07:
i assume you wiped it using the st-link because dfuse flashing 1.1 followed by 1.3 did not improve things?

bdz69, 2018-05-07:
Yes St-Link V2 JTAG programmer. I tried it first with the standard firmware 1.1 and that wouldn’t work. Symptoms were exactly as you described. Laurent sent me a knew firmware 1.1 that was made specially to fix the registers that were changed with the 1.3 version. The tags I flashed after complete erase with the ST-Link worked immediately with all the LEDs flashing like they used to when communicating.

lpdunwell, 2018-05-14:
thanks for replacing the damaged equipment!

however, after spending ~5h trying to get basically anything to work using the windows gui,
i repacked all of it and shelved it.

nothing works as stated, i get all kinds of unrelated and 'unexpected' errors, and i think 'settings don't get written to all the devices in range' with range >= 3m just kinda prevents much success anyway.