pozyxLabs/Pozyx-Arduino-library #75: Library seem to be incompatible with PJRC Teensy boards

nesdnuma, 2018-06-05:
When trying to compile the examples for a Teensy 3.1 board, I get a lot of error messages (see below).

It's too bad as the Teensy can communicate much faster with a computer than a regular Arduino Uno.

_In file included from unit_tests_core.ino:4:0:

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:449:90: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int getRead(uint8_t reg_address, uint8_t *pData, int size, uint16_t remote_id=NULL);

                                                                                          ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:463:91: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int setWrite(uint8_t reg_address, uint8_t *pData, int size, uint16_t remote_id=NULL);

                                                                                           ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:479:141: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int useFunction(uint8_t reg_address, uint8_t *params=NULL, int param_size=0, uint8_t *pData=NULL, int size=0, uint16_t remote_id=NULL);

                                                                                                                                             ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:557:76: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int getLastNetworkId(uint16_t *network_id, uint16_t remote_id = NULL);

                                                                            ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:570:77: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int getLastDataLength(uint8_t *data_length, uint16_t remote_id = NULL);

                                                                             ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:595:71: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int setNetworkId(uint16_t network_id, uint16_t remote_id = NULL);

                                                                       ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:608:82: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int getUWBSettings(UWB_settings_t *UWB_settings, uint16_t remote_id = NULL);

                                                                                  ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:623:82: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int setUWBSettings(UWB_settings_t *UWB_settings, uint16_t remote_id = NULL);

                                                                                  ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:638:90: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int setUWBSettingsExceptGain(UWB_settings_t *UWB_settings, uint16_t remote_id=NULL);

                                                                                          ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:651:68: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int setUWBChannel(int channel_num, uint16_t remote_id = NULL);

                                                                    ^

D:\Programmation\Arduino\libraries\Pozyx/Pozyx.h:666:69: warning: converting to non-pointer type 'uint16_t {aka short unsigned int}' from NULL [-Wconversion-null]

     static int getUWBChannel(int* channel_num, uint16_t remote_id = NULL);

[... 532 lines cut ...]
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:134:11: note:   candidate expects 1 argument, 2 provided

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:135:11: note: candidate: String& String::concat(long unsigned int)

  String & concat(unsigned long num)  {return append(num);}

           ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:135:11: note:   candidate expects 1 argument, 2 provided

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:136:11: note: candidate: String& String::concat(float)

  String & concat(float num)   {return append(num);}

           ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:136:11: note:   candidate expects 1 argument, 2 provided

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:137:11: note: candidate: String& String::concat(double)

  String & concat(double num)   {return append(num);}

           ^

C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy3/WString.h:137:11: note:   candidate expects 1 argument, 2 provided_

laurentva, 2018-07-05:
Hello,

The Pozyx functions are only giving you warnings. The one error you're getting is this:
`D:\Programmation\Arduino\libraries\ArduinoUnit\src\ArduinoUnitUtility\ArduinoUnitMockPrint.cpp:136:95: error: no matching function for call to 'MockPrint::concat(const char*, size_t&)'`

So I'd say the ArduinoUnit (https://www.arduinolibraries.info/libraries/arduino-unit) library is giving you issues for Teensy. We use it for our unit tests.

What I'd do if I were you is 
* fork this library and clone it locally as the used Teensy library
* remove all unit tests and references to ArduinoUnit
* ??? (test)
* profit!

I'll see if I can make the removal of the ArduinoUnit for the actually published library a long-term goal.

Kind regards,

closed 2018-07-05 by laurentva

GOBish, 2018-07-14:
Any progress on this issue? I am a big fan of the Teensy boards and would like to stick with them.
Paul from PJRC made this library update a while ago to work with Pozyx but I don't know how well it works or if includes the latest updates from Pozyx:
https://github.com/PaulStoffregen/Pozyx-Arduino-library

nesdnuma, 2018-07-15:
Actually I could modify Pozyx latest lib's version according to the advice on PJRC forum and compile it for the Teensy (using the Sparkfun Teensy to Uno shield adapter). When it functions, it's much faster than with a Uno board (I could go down to a latency of 6ms) but it does not always start when opening the serial port. I couldn't find the cause of this issue yet. Any suggestion?

GOBish, 2018-07-15:
Thanks for the reply. That would be great if you could modify the library. Not sure what you mean by "it does not always start when opening the serial port" - could you explain further?

nesdnuma, 2018-07-15:
I mean that sometimes it functions sometimes it doesn't whereas on the Uno it always functions.

GOBish, 2018-07-15:
Here are a couple of links where the issue was discussed:

https://forum.pjrc.com/threads/52648-Pozyx-library-issue

and here:
https://forum.pjrc.com/threads/47047-Position-recognized-audio-player

tonton81, 2018-07-15:
specifically when you say sometimes it works and sometimes it doesn't, do you mean the serial port doesn't open? If thats the case, just add a little delay in setup() before your code for serial hardware to initialize (delay(1000)), this should make legacy comms readers work with the fast startup of teensy

GOBish, 2018-07-15:
@nesdnuma - could you post your modified code please?

nesdnuma, 2018-07-16:
@GOBish:

I think that Paul used a library version which is a bit old whereas I've used latest version.

I am not sure about which files from the Pozyx lib have been modified. Actually I didn't think I would post them here because the first answer I got from Pozyx support about the Teensy was that it was not supported. I am glad to see it might have evolved.

Basically you add this code in Pozyx_core.cpp:
```
#elif defined(__arm__) && defined(TEENSYDUINO)
    // Teensy LC & 3.x
    attachInterrupt(interrupt_pin+2, IRQ, RISING);
```
just before
```
#else
    Serial.println("This is not a board supported by Pozyx, interrupts may not work");
    attachInterrupt(interrupt_pin, IRQ, RISING);
#endif
```

Then you replace any occurence of "NULL" with a 0 (zero) in all the files (NULL is not supposed to be used anymore in Arduino IDE I think).

@tonton81: the `delay(1000)` after serial port opening seem to improve chances of success but I do not have much time to run further tests today.

PaulStoffregen, 2018-07-31:
> I mean that sometimes it functions sometimes it doesn't whereas on the Uno it always functions.

@nesdnuma - This is probably because opening the port causes Arduino Uno to reset.  But many newer boards with native USB do not reset simply by opening the port.  If they boot up quickly, the Serial.print() will already have sent data before you managed to open the serial monitor.  The 1000 ms delay "works" if you manage to open the serial monitor in that 1 second.

Most modern Arduino examples now use this to start serial communication:

    //Initialize serial and wait for port to open:
    Serial.begin(9600);
    while (!Serial) {
      ; // wait for serial port to connect. Needed for native USB port only
    }

Here's a link to one of Arduino's examples using this:

https://github.com/arduino/Arduino/blob/master/build/shared/examples/04.Communication/ASCIITable/ASCIITable.ino

I recommend copying this style, so things work with for Teensy and the many newer Arduino boards which also need this: MKR1000, Zero, Leonardo, Micro, etc.

nesdnuma, 2018-08-07:
This is really something everybody really wants to know, Paul! This might explain some issues I've had with Micro and Teensy for years! Thanks a lot.